arduino/mcuboot-arduino-stm32h7 #14: Don't clear Reset Source Register on boot

mkj, 2022-05-10:
This bootloader (and the standard Portenta h7 one) clears the RSR so it isn't possible for subsequent programs to check the reset reason. Could it be changed to keep the reset source register intact? Thanks.

https://github.com/arduino/mcuboot-arduino-stm32h7/blob/a828baf73f93b6aa855a5d185f0ca4fed5755e2f/app/main.cpp#L229

pennam, 2022-05-10:
Hi @mkj, thanks for the report.
I've just had a quick look and i think that it can be done calling directly the mbed-os hal function from the bootloader. I will give it a try and let you know.

pennam, 2022-05-25:
@mkj storing the reset reason somewhere else, like in an RTC register, would be an acceptable solution?

wero1414, 2024-03-07:
@pennam any news on this issue? i have a board getting rebooted, and i would like to address the reset cause