arduino/mcuboot-arduino-stm32h7 #25: Cannot build with current version of mbed OS

doesnotcompete, 2023-01-19:
Hello,
thanks for making this bootloader publicly available!
Unfortunately, the mbed distribution specified in `mbed-os.lib` does not seem to be available anymore. Building with a current ARM-distributed version of mbed causes an issue during linking:
```
Link: mcuboot-arduino-stm32h7
/usr/lib/gcc/arm-none-eabi/12.2.0/../../../../arm-none-eabi/bin/ld: BUILD/PORTENTA_H7_M7/GCC_ARM/mbed-os/storage/filesystem/fat/source/FATFileSystem.o: in function `mbed::FATFileSystem::format
(mbed::BlockDevice*, unsigned long long)':
/home/kevin/Development/ipt/mcuboot-arduino-stm32h7/./mbed-os/storage/filesystem/fat/source/FATFileSystem.cpp:420: undefined reference to `f_mkfs'
collect2: error: ld returned 1 exit status
[ERROR] /usr/lib/gcc/arm-none-eabi/12.2.0/../../../../arm-none-eabi/bin/ld: BUILD/PORTENTA_H7_M7/GCC_ARM/mbed-os/storage/filesystem/fat/source/FATFileSystem.o: in function `mbed::FATFileSystem
::format(mbed::BlockDevice*, unsigned long long)':
/home/kevin/Development/ipt/mcuboot-arduino-stm32h7/./mbed-os/storage/filesystem/fat/source/FATFileSystem.cpp:420: undefined reference to `f_mkfs'
collect2: error: ld returned 1 exit status
```
This seems to be due to the option `fat_chan.ff_use_mkfs` being disabled. On the other hand, building with `mkfs` enabled makes the build too big to fit in the designated memory area.
Can you give a hint as to how I can build the bootloader? I am trying to flash a custom MCUBoot image to the slot at `0x8020000`, but this always leaves my Portenta H7 in a state where e.g. the PMIC is still configured correctly when power is connected, but nothing else happens. `gdb` indicates the board is stuck somewhere in the bootloader's address space, and I am hoping getting some logging from the bootloader could help here.

pennam, 2023-01-19:
Hi @doesnotcompete 
this branch should work for portenta: https://github.com/arduino/mbed-os/tree/extrapatches-6.16.0

Bootloader have to be be flashed at `0x8000000`. Without configuring MCUboot keys in flash it will behave like a normal bootloader trying to boot a sketch from `0x8040000`. If you want to make it boot an MCUboot image you need to flash the keys: to do that you can take a look to this sketch:

https://github.com/arduino/ArduinoCore-mbed/blob/781960862fbddeee35c3de4fb3386eea08a7934e/libraries/STM32H747_System/examples/STM32H747_manageBootloader/STM32H747_manageBootloader.ino#L338 

The sketch will also let you decide which bootloader standard/MCUboot load on your portenta and if load or not the keys.

After you flash the keys the MCUboot bootloader will start booting MCUboot images from address `0x8020000`. At this point Images have to be signed with the correct key otherwise they won't boot. If you flash the MCUBoot bootlader version included in the example sketch and the default keys you can produce signed and encrypted MCUboot images with the IDE selecting `signature + encryption` from the Tools -> Security settings menu; rr with the arduino-cli following this guide https://arduino.github.io/arduino-cli/0.29/guides/secure-boot/

pennam, 2024-05-06:
fixed by https://github.com/arduino/mcuboot-arduino-stm32h7/commit/370dce28b648aea24b7ae2a0f685a59e5439f672

closed 2024-05-06 by pennam