arduino/mcuboot-arduino-stm32h7 #31: Can not build in Mbed Studio terminal

iossocket, 2024-02-06:
Hello,

When I build mcuboot-arduino-stm32h7 in Mbed Studio following the README. Encounter an error:

```
➜  mcuboot-arduino-stm32h7 git:(main) mbed compile -m PORTENTA_H7_M7 -t GCC_ARM --profile=release --profile custom.json
[mbed] Working path "/Users/sashimi/Mbed Programs/mcuboot-arduino-stm32h7" (program)
Could not find executable for GCC_ARM.
Currently set search path: /Library/Application Support/Mbed Studio/mbed-studio-tools/gcc/bin
[mbed] ERROR: "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/bin/python" returned error.
       Code: 2
       Path: "/Users/sashimi/Mbed Programs/mcuboot-arduino-stm32h7"
       Command: "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/bin/python -u /Users/sashimi/Mbed Programs/mcuboot-arduino-stm32h7/mbed-os/tools/make.py -t GCC_ARM -m PORTENTA_H7_M7 --profile release --profile custom.json --source . --build ./BUILD/PORTENTA_H7_M7/GCC_ARM-RELEASE"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```
Debug into the source code `mcuboot-arduino-stm32h7/mbed-os/tools/make.py`. I found the error was thrown from `toolchain_names = get_valid_toolchain_names(target, toolchain)`, which is defined in `build_api.py`. So I suspect I missed some toolchain bin or used the wrong mbed cli? 

Appreciate some one can help me out.

### Environment
OS: MacOS
Mbed Studio: 1.4.5.19
mbed cli: 1.10.4

pennam, 2024-02-09:
> Could not find executable for GCC_ARM.

@iossocket you should  set the default toolchain

`mbed config -G GCC_ARM "path to your toolchain"`

iossocket, 2024-02-11:
> > Could not find executable for GCC_ARM.
> 
> @iossocket you should set the default toolchain
> 
> `mbed config -G GCC_ARM "path to your toolchain"`

Hi @pennam, thanks for your reply.  After configuring the global GCC_ARM path as "/Library/Application Support/Mbed Studio/mbed-studio-tools/gcc/bin" which is the installed by *Mbed Studio*, the compile command thrown the the same error. 

Inside the "/Library/Application Support/Mbed Studio/mbed-studio-tools/gcc/bin" folder, I can see two files: *arm-none-eabi-gdb* and *arm-none-eabi-objdump*. Dose that mean I need some extra executable file in the bin folder? 

Thanks :)

pennam, 2024-05-06:
@iossocket i've lost your last message, sorry. Did you manage to build the bootloader in Mbed Studio terminal?