arduino/mcuboot-arduino-stm32h7 #32: imgtool crash when using example in readme.md

ddmesh, 2024-04-12:
When using the example to create the output.bin file, imgtool crashes.

Command
~~~sg
 imgtool sign --key ecdsa-p256-signing-priv-key.pem --encrypt ecdsa-p256-encrypt-pub-key.pem input.bin output.bin --align 32 --max-align 32 --version 1.2.3+4 --header-size 0x20000 --pad-header --slot-size 0x1E0000
~~~
Error:
~~~
/github-mcuboot/scripts/imgtool/image.py", line 568, in add_header
    header = struct.pack(fmt, struct.error: ushort format requires 0 <= number <= (0x7fff * 2 + 1)
~~~

When I use a header-size of 0x2000 instead of 0x20000, the tool is working. But it is unclear, if 2000 is correct or not.

pennam, 2024-04-12:
@ddmesh which imgtool version are you using? You should use the patched version that comes in this branch https://github.com/arduino/mcuboot/tree/stm32h7-1.8.0

ddmesh, 2024-04-14:
Hi have used this version. please check if the header size in the example I mentioned is correct. When header is created, it expects a 16bit value. 0x20000 does not fit into this.
Issue relates to this this main git branch.

pennam, 2024-05-06:
Increasing the maximum header size is one of the changes we have made in order to support our flash layout:

https://github.com/arduino/mcuboot/commit/52dbe250a8d3c5cf204bda43737cc7eb0fc552f2

could you please doublecheck you are on the correct branch/commit?

ddmesh, 2024-05-08:
hi. I used the latest version (head of main branch). please note that the 0x20000 never would fit into 16bits. the python script complains about it and throws an error.
either you should use 0x2000 instead 0x20000 or you need to change a lot in bootloader and imgtool. ;-)

pennam, 2024-05-08:
That is exactly what we have done, you don't have to use the `main` branch, but the `stm32h7-1.8.0` branch of this forked repo `https://github.com/arduino/mcuboot`

ddmesh, 2024-05-08:
hi. this was not clear to me that there is another repository because everywhere this one here is referenced. so when this repository is deprecated and should not be used, then references at Arduino documents must be updated and this repo should be closed or at least point to new mcuboot Arduino repo. it is really confusing, finding correct versions ;-)
I will check the new repository. if you think the issue is solved, you may close it.
but this repo here. has this bug still unsolved. it is advisable to close this repo if development has continued elsewhere. other people will run in same issues in future.

pennam, 2024-05-09:
Hi, this repo is not deprecated or outdated, it contains code that should be used to build a bootloader that can act as a "standard" Arduino bootloader or as a "secure bootloader" based on mcuboot code.

Following the readme after cloning this repo you have to download the `mbed-os` and `mcuboot` git submodules with the following command

```
mbed config root . && mbed deploy
mbed compile -m PORTENTA_H7_M7 -t GCC_ARM --profile=release --profile custom.json
```

this will download the correct version of `mbed-os` and `mcuboot` repository

```
$ mbed config root . && mbed deploy
[mbed] . now set as default root in program "mcuboot-arduino-stm32h7"
[mbed] Working path "/tmp/mcuboot-arduino-stm32h7" (program)
[mbed] Adding library "mcuboot" from "https://github.com/arduino/mcuboot" at rev #f53a4d0c7b86
[mbed] Adding library "mbed-os" from "https://github.com/arduino/mbed-os" at rev #2e1da01300f1
```

closed 2024-05-09 by pennam