arduino/mcuboot-arduino-stm32h7 #33: loading update image via DFU to external flash stucks (Finder OPTA)

ddmesh, 2024-04-12:
When following the example in readme.md [Load an update sketch](https://github.com/arduino/mcuboot-arduino-stm32h7?tab=readme-ov-file#load-an-update-sketch) for OPTA device from Finder, the upload get stuck.

The external QSPI Flash is formated as follows (from examples):
~~~cpp
std::vector<Partition> partitioningScheme  =
{
         {1*1024, FS_FAT}, // 1 MB for certificates
         {5*1024, FS_FAT}, // 5 MB for OTA firmware updates
         {8*1024, FS_LITTLEFS} // 8 MB for user data
};
~~~

dfu-utils -l:
~~~sh
Found DFU: [2341:0364] ver=0200, devnum=24, cfg=1, intf=0, path="3-2.3.2.4.1", alt=3, name="@Arduino  boot  v.25   /0x00000000/0*4Kg", serial="003400303132511639313731"
Found DFU: [2341:0364] ver=0200, devnum=24, cfg=1, intf=0, path="3-2.3.2.4.1", alt=2, name="@Ext File Flash  0MB   /0x00000000/0*4Kg", serial="003400303132511639313731"
Found DFU: [2341:0364] ver=0200, devnum=24, cfg=1, intf=0, path="3-2.3.2.4.1", alt=1, name="@Ext RAW  Flash 16MB   /0x90000000/4096*4Kg", serial="003400303132511639313731"
Found DFU: [2341:0364] ver=0200, devnum=24, cfg=1, intf=0, path="3-2.3.2.4.1", alt=0, name="@Internal Flash  2MB   /0x08000000/01*128Ka,15*128Kg", serial="003400303132511639313731"
~~~

flash command:
~~~sh
dfu-util --device 0x2341:0x035b -D update.ino.bin -a2 --dfuse-address=0xA0000000:leave
~~~
Tested with given address 0xA0000000 and also with the one given by dfu-util 0x00000000.

The flash stucks at 0% and the DFU MCUBoot is blocked afterwards.

pennam, 2024-05-06:
Hi @ddmesh secure upload of the sketck on the external QSPI flash is enabled only if you have configured the keys in your board. You can do it using this sketch:

 - https://github.com/arduino/ArduinoCore-mbed/blob/main/libraries/MCUboot/examples/enableSecurity/enableSecurity.ino
 
 Load the sketch, open the serial monitor and answer `Y` to the questions. This will upload default keys in the internal flash. You can also customize the key, generating yours and substitute them in the Sketch `ecdsa-p256-encrypt-key.h` and `ecdsa-p256-signing-key.h` files. When the process ends power cycle you OPTA, at reboot red-blue LED will blink for a while and then green LED will start to fadein-fadeout. Now your board is ready to upload your secure sketch on the external QSPI flash. You can do this with the command you have already used or with the IDE enabling Security settings Signature + Encryption in the Tools menu.
 
 If you want to go back to the "standard" booloader working mode you need to reflash the bootloader using a SWD programmer. So make sure to have one on your desk.

schnoberts1, 2025-03-26:
... for the record I go back and forth between secure and non-secure modes using dfu-util on the Giga. It seems quite happy to take a new bootloader and forget its keys.

pennam, 2025-03-27:
> ... for the record I go back and forth between secure and non-secure modes using dfu-util on the Giga. It seems quite happy to take a new bootloader and forget its keys.

@schnoberts1 are you enabing dfu mode using the boot0 button?

schnoberts1, 2025-03-27:
Yes.