arduino/mcuboot-arduino-stm32h7 #34: How do I build this?

schnoberts1, 2024-08-22:
It seems to be dependent on a repo I can't access, see last line.

```
(base) andy@ice mcuboot-arduino-stm32h7 % mbed config root . && mbed deploy
[mbed] . now set as default root in program "mcuboot-arduino-stm32h7"
[mbed] Working path "/Users/andy/code/dbc/g/3rdparty/mcuboot-arduino-stm32h7" (library)
[mbed] Program path "/Users/andy/code/dbc/g/3rdparty/mcuboot-arduino-stm32h7"
[mbed] WARNING: Program "mcuboot-arduino-stm32h7" in "/Users/andy/code/dbc/g/3rdparty/mcuboot-arduino-stm32h7" does not use source control management.
       To fix this you should use "mbed new ." in the root of your program.
---
[mbed] Adding library "mcuboot" from "https://github.com/arduino/mcuboot" at rev #f53a4d0c7b86
[mbed] Adding library "mbed-os" from "https://github.com/bcmi-labs/mbed-os" at rev #a55c905a33d0
Username for 'https://github.com': ^C[mbed] User aborted!
(base) andy@ice mcuboot-arduino-stm32h7 %
(base) andy@ice mcuboot-arduino-stm32h7 %
(base) andy@ice mcuboot-arduino-stm32h7 % mbed config root . && mbed deploy
[mbed] . now set as default root in program "mcuboot-arduino-stm32h7"
[mbed] Working path "/Users/andy/code/dbc/g/3rdparty/mcuboot-arduino-stm32h7" (library)
[mbed] Program path "/Users/andy/code/dbc/g/3rdparty/mcuboot-arduino-stm32h7"
[mbed] WARNING: Program "mcuboot-arduino-stm32h7" in "/Users/andy/code/dbc/g/3rdparty/mcuboot-arduino-stm32h7" does not use source control management.
       To fix this you should use "mbed new ." in the root of your program.
---
[mbed] Updating library "mcuboot" to rev #f53a4d0c7b86
[mbed] Adding library "mbed-os" from "https://github.com/bcmi-labs/mbed-os" at rev #a55c905a33d0
Username for 'https://github.com':
```

schnoberts1, 2024-08-23:
Ok, I see that this was a bug in RC2 that’s been fixed.

I assume therefore that main is legit and I can live off the HEAD commit.

pennam, 2024-08-23:
@schnoberts1 yes you should be able to build this from HEAD. Let me know if it does not work.

schnoberts1, 2024-08-23:
Thanks for the quick response.

closed 2024-08-23 by schnoberts1

schnoberts1, 2024-08-27:
So it would seem to work but I notice that post build it's patched the following files:
```
	modified:   variants/GENERIC_STM32H747_M4/cflags.txt
	modified:   variants/GENERIC_STM32H747_M4/cxxflags.txt
	modified:   variants/GENERIC_STM32H747_M4/defines.txt
	modified:   variants/GENERIC_STM32H747_M4/linker_script.ld
	modified:   variants/GIGA/defines.txt
	```

The patch to the linker script breaks RPC because it removes the openamp section. The others don't seem to make a difference but I'm wary of differences. 

We get `-fno-exceptions` in the c/cxxflags. I'm cool with this, I don't like with sup++.

It resets CORE_MAJOR etc

```
diff --git a/variants/GENERIC_STM32H747_M4/defines.txt b/variants/GENERIC_STM32H747_M4/defines.txt
index 1da443e9..e9227958 100644
--- a/variants/GENERIC_STM32H747_M4/defines.txt
+++ b/variants/GENERIC_STM32H747_M4/defines.txt
--DCORE_MAJOR=4
--DCORE_MINOR=1
--DCORE_PATCH=5
+-DCORE_MAJOR=
+-DCORE_MINOR=
+-DCORE_PATCH=
```

That seems less awesome tbh.