arduino/mcuboot-arduino-stm32h7 #37: Query regarding use on the Giga WiFI R1 along with OTA

schnoberts1, 2025-03-24:
Hi,

I can see this repo supports the Giga but is this the case for the documentation. i.e. 

```
dfu-util --device 0x2341:0x035b -D update.ino.bin -a2 --dfuse-address=0xA0000000:leave
```
Is this expected to work on the Giga once keys have been flashed into place?  How does this interact with OTA where there's a requirement for FAT formatted partition? Does the above trash any partitioning?

schnoberts1, 2025-03-24:
I guess this is the answer 

const uint32_t QSPIFLASH_BASE_ADDRESS   =  0x90000000;
const uint32_t FILEBLOCK_BASE_ADDRESS   =  0xA0000000;

closed 2025-03-24 by schnoberts1

pennam, 2025-03-24:
Hi @schnoberts1 

> Does the above command trash any partitioning and formatting done?

The dfu-util command will preserve the FAT partition, the command will create a UPDATE.BIN file inside the 2nd partition of the QSPI flash. 

> I guess I want to know if I can upload directly to the device, deploy it and upload via OTA or whether I have to do some QSPI reformatting post-local deployment?

The 2nd partition is present on all the boards as per default factory partitioning; as long as you don't change the partition table of the QSPI flash you don't need any reformatting

> Finally, can I upload signed and encrypted bin files via arduino-cli and skip the interim stem of loading to QSPI first?
Yes it should be possible using the -b switch

`-b arduino:mbed_portenta:envie_m7:security=sien`

schnoberts1, 2025-03-24:
Hi,I closed the ticket having read the boot loader code and realising what it’s doing with this 0xA address. All good!