arduino/mcuboot-arduino-stm32h7 #38: Header size for imgtool sign in the documentation is set to 0x20000

schnoberts1, 2025-03-24:
Hi, 

This seems quite large and also outside the range of the pack format in image.py. Result:

```
    self.add_header(enckey, protected_tlv_size)
  File "/home/andy/code/dbc/test/tools/provisioning/.venv/lib/python3.12/site-packages/imgtool/image.py", line 573, in add_header
    header = struct.pack(fmt,
             ^^^^^^^^^^^^^^^^
struct.error: 'H' format requires 0 <= number <= 65535
```

Using 0x2000 (assuming the doc had an extra zero) doesn't result in an error. I've not tested the upload yet.

schnoberts1, 2025-03-24:
So it turns out that the imgtool shipped with Arduino is different from the standard one. The Arduino one accepts the large header value. Whether this is right, I don't know. Certainly the resulting image won't upload to my Giga. Probably error between chair and keyboard so I'll close this particular ticket though I would ask that your documentation specifies what tooling to be used, imgtool and dfu-util are pretty standard tools.

closed 2025-03-24 by schnoberts1

pennam, 2025-03-24:
@schnoberts1 this is not a typo, we had to do this to keep the arduino sketch start address to `0x08040000` and the start of the header aligned to the flash sector @ `0x08020000`

schnoberts1, 2025-03-24:
Can I grab the custom imgtool from a repo?

schnoberts1, 2025-03-24:
Awesome, thnx.

  

https://github.com/arduino/mcuboot/tree/stm32h7-1.8.0/scripts