arduino/mcuboot-arduino-stm32h7 #42: Question: How is M4 supported with secure boot

schnoberts1, 2025-03-26:
I'm running this:

```
dfu-util -w --device 0x2341:0x035b -D "${bin_file}" -a2 --dfuse-address=0xA0000000:leave
```
where bin_file is a path of a `.bin` file. 

When the bootloader responds to the dfu request does it end up running `tryOTA()`? I see that inspects the first word. The first word for my M7 compiled bins is 0000 and 8000 for M4. Does that seem correct?

pennam, 2025-03-26:
M4 is not supported by mcuboot OTA out of the box, it should theoretically work by combining M7 and M4 binaries and then sign the combied binary with imgtool, but i never tried myself.

mcuboot OTA and Arduino standard OTA are working in two different ways:

* Arduino OTA
put UPDATE.BIN file you your QSPI filesystem
write some magic data in RTC backup registers
reset the board
...
At boot Arduino bootloader will
...
check secure boot is disabled
check backup registers
check the presence of UPDATE.BIN file
call tryOTA()

* MCUboot OTA
put UPDATE.BIN file you your QSPI filesystem
write some magic in the mcuboot image TLV
reset the board
....
At boot Arduino Bootloader will
...
check secure boot is enabled
start mcuboot
mcuboot will take care of the image [swap](https://github.com/arduino/mcuboot/blob/f53a4d0c7b866e0d98062d83b5a8a8e0dc20099c/boot/bootutil/src/swap_scratch.c#L676)

schnoberts1, 2025-03-26:
I guess I’m not sure of the difference. I’m using the bootloader build from this repo. Is this mcuboot OTA or Arduino OTA or is it doing both, mcuboot via the dfu interface and Arduino OTA using the backup registers? I’m a bit confused tbh.

pennam, 2025-03-26:
Is doing both, depending whether or not you have written encryption and signing keys in the board flash.

schnoberts1, 2025-03-26:
… or maybe you are saying that secure boot gets you Arduino OTA but if I want signed and encrypted firmware it is via mcuboot OTA?

pennam, 2025-03-26:
You have 2 possibilities:

Signing and encryption keys are not in the flash -> Arduino bootloader and Arduino OTA (tryOTA)
Signing and encryption keys are in the flash -> mcuboot bootloader and mcuboot OTA with swap and resume; signed and encrypted images.

pennam, 2025-03-26:
In case of Arduino OTA with tryOTA M4 update will work only if you are using the 50/50 flash split because of [this](https://github.com/arduino/mcuboot-arduino-stm32h7/blob/370dce28b648aea24b7ae2a0f685a59e5439f672/app/ota/ota.cpp#L37) hardcoded address

schnoberts1, 2025-03-26:
Easy to fix :)

schnoberts1, 2025-03-26:
If “ Signing and encryption keys are not in the flash” then this means no secure firmware unless I pre-decrypt and validate prior to writing to QSPI?Is there some docs describing this detail? I feel like I have missed a doc here describing the trade offs :)

pennam, 2025-03-26:
yes this means no secure firmware and no mcuboot. so in that case you can't upload the update image directly on the QSPI flash with `dfu-util -w --device 0x2341:0x035b -D "${bin_file}" -a2 --dfuse-address=0xA0000000:leave`
the -a2 switch is only for mcuboot images

schnoberts1, 2025-03-26:
Thanks for you help. I'll go read the code more thoroughly now and try out the both bins in one bin approach and see if that works.

schnoberts1, 2025-03-27:
Ok, I opted for the mcuboot approach. Using a composite bin file containing M7 and M4 images (with the M4 at its offset as it would be in the firmware) worked great. Most of the padding compresses away if needed and the ability to upgrade both cores at the same time is much simpler than one by one. Thanks for that @pennam . The whole swap process is slow, firmware updates are in the order of 3 minutes for my 1.5MB encrypted signed .bin file.

There were two key things I had address to get here (for others on this path). 

1. You can't instantiate a QSPIBlockDevice like shown in the examples throughout the code base, you need to use default_instance.
2. I am using LittleFilesystem2 and that had a bug in the constructor that crashed unmounts when the device hadn't been mounted (i.e. when you reformat a device without first mounting the filesystem). 

The actual arduino mcuboot loader itself worked fine once I understood the trade offs you identified.

closed 2025-03-27 by schnoberts1

schnoberts1, 2025-03-27:
> Thank you so much for your feedback @schnoberts1
> Thank you so much for your feedback @schnoberts1 <https://github.com/schnoberts1>

… just one note, since then, through a bunch of dev style stress testing where I’m uploading stuff, interrupted in, resetting the device (pressing RST) etc I have got the device in a state where it no longer accepts firmware downloads. Firmware is downloaded, then rather than move into the swap mode and blink purple it just sticks with the back led on green for a while. No amount of QSPI reformatting etc etc has resolved this so I’m going to file this under not reliable enough for the field for the time being

pennam, 2025-03-28:
Are you able to check what is the output on the `Serial1` debug console?

schnoberts1, 2025-03-28:
I'll get that sorted next week and I’ll report back.